AeolusUX/ArkOS-R3XS #125: Problem with 128 GB micro SD installation

FireStrings, 2025-04-17:
Hey guys!

I was trying to install the ArkOS 2.0 (03302025) in my 128GB micro SD, but I was having trouble, a different type of black screen: turning on the device, it was showing the ArkOS boot logo normally, but after ~1second, the black screen appeared and then remained forever, forcing me to turn off directly on the power on/off button.

Just to be sure about the micro SD quality / some windows 10 issue / the ArkOS image failure or something that could be causing it, I just decided to test the micro SD from my phone, a 8GB miro SD. And then....ta-dah! It works. So, I concluded that the problem its the size of the card.

Searching more, I've found a [post](https://www.reddit.com/r/R36S/comments/1e4n4kf/help_flashing_new_sd_card_with_arkos_im_losing_my/?rdt=59296) that its related to a similar problem. So, is it a tracked issue? Do you plans to work on it?

Anyways, maybe you can put something on the wiki warning about this SD size problem, if it's not there yet. 

Thank you!

requium94, 2025-04-18:
Made an account just to comment because I recently solved a similar issue. I was able to update to the latest arkos on the SD card the device came with, but when it came to updating a 256gb Samsung card, it would seem to succeed in writing the image but the Easyroms partition didn't appear and the device wouldn't install the image. 

Turned out my issue was having a cheap usb3.0 SD reader that I got from eBay for $8.00AUD and getting a Lexar branded one from a brick and mortar store sorted my issue immediately.

That said I believe it's known by the community that many SD cards and brands aren't compatible, hence why I bought Samsung. Here's a compatibility list:

https://docs.google.com/spreadsheets/u/0/d/1gWxtr-GmwWop-_qGUq022RXxK2aTLpPg9Qra68TQLI8/htmlview#gid=0

FireStrings, 2025-04-20:
Well, the last tests were:

- Tried to flash a Sandisk 128GB White & Gray, with a black screen after the ArkOS logo
- Tried to flash a Sandisk 64GB White & Gray, with a black screen after the ArkOS logo
- Tried to use the two above only as ROMS folder (formatted as exFAT), in the second slot, but the roms was "disappearing", even with the sd card taking up space, as if they were there (it was weird).

According to the spreadsheet shared, the mentioning sd card its compatible. Do you guys are seeing something that I made wrong here?

FireStrings, 2025-04-21:
Hey @requium94 , thanks for the answer!

I did this tutorial again, and waited approximately ~2h, with no success. I had already done this tutorial a dozen times, with more than one card. But the problem does not seem to be the classic black screen, but rather a black screen after the ArkOS logo appears on boot (that is, I start the console with the SD in slot 1, the ArkOS logo appears, and then a black screen appears).

As I mentioned, the only SD card that worked was an old Sandisk, 8GB. I tested it with another Sandisk card, 64G and 128G, both with the same behavior (by the way, I already checked the spreadsheet, they are compatible, because they are Ultra Class 10 (White & Gray)). This is talking about the flash to install the operating system.

After I managed to install the system on the 8GB card, another problem is that I can't even use the other two Sandisk cards as the ROM directory (in the second slot, like SD2), because when I try to compile the ROMs, while the card is in the PC, all the files are there. When I put it in the console, it says there are no ROMs at all, or only a small fraction. And when I put the card back in the PC, the ROMs are really gone (and the strangest thing is that the space on the card remains as if the ROMs were there, but they aren't). And after all this, when I try to copy any file to the card, the file simply disappears.

So, I'm convinced that there isn't much compatibility between the console and the new Sandisk cards.

WDYT?

requium94, 2025-04-22:
Don't think I can be of much help I'm afraid. I had the sd reader pictured and it only writing to a 64gb no brand card, but wouldn't write to a new Samsung card. If you trust your sd reader and don't ahve a second one to try with, not sure whats going on.

[image]

FireStrings, 2025-04-22:
Well, imagine that could also be it, because I can't think of any other alternatives lol

FireStrings, 2025-04-24:
Hey folks, just an update: Yesterday I gave up on Sandisk. Checking more detailed, I noticed that the sd card was exhibiting a strange behavior: not formatting in specific file systems, files disappearing, the poor reading/writing speed, the flash problem itself. All of this make me think that its a fake card. Reading articles about it, I could see that Sandisk has a high rate of falsification.

Well, now with a Samsung Evo 128GB, the problems are over, the sd card worked well. So, if there was no testes with a real Sandisk sd, I dont know if make sense to keep the "Some reports say that some batches of this doesn't work either. So I'd just stay away." in [the spreadsheet](https://docs.google.com/spreadsheets/u/0/d/1gWxtr-GmwWop-_qGUq022RXxK2aTLpPg9Qra68TQLI8/htmlview#gid=0), cause the chance of the card being fake is high.

AeolusUX, 2025-04-25:
> Hey folks, just an update: Yesterday I gave up on Sandisk. Checking more detailed, I noticed that the sd card was exhibiting a strange behavior: not formatting in specific file systems, files [...]

I'd like to keep it there since even if there are tons of fakes and over 3/4 of the listings on the internet are fakes Sandisk is still one of the most trusted brands out there for SD Cards. ( Also explains why it's being faked so much).

closed 2025-05-14 by AeolusUX