AeolusUX/ArkOS-R3XS #129: ArkOS R3XS V2.0 (03302025) R36S pcsx rearmed enhanced resolution setting causes flickering

mrmcsirguy, 2025-04-24:
Hi. I have an R36S with a fresh install, and everything works great except for this one thing. Enabling enhanced resolution causes a black frame or random previous frame to flicker together with the current PSX game display. I adjust all settings and tried changing bios to no avail. Also, the standard bios like scph5501.bin are reported but psxonpsp660.bin never gets loaded. Any way to fix this issue?

ZerbZerb, 2025-05-12:
Hi, I just want to confirm on here, that I tested various settings and setups and can confirm that the lastes update in March is causing the flickering in Pcsx Rearmed enhanced resolution settings.

It would be great if this could be fixed!

ZerbZerb, 2025-05-26:
Hi, I wanted to add that even in the latest update of May, the flickering at enhanced resolution, is still there for many psx games.

doubleyoustew, 2025-06-06:
Same here. Enabling Enhanced resolution causes flickering in games like GT1&2 and THPS2. Tekken 3 works fine.

AeolusUX, 2025-06-10:
This is something that comes with an updated retroarch version so it could be an issue with retroarch itself which , libretro would need to address.

closed 2025-06-10 by AeolusUX