AeolusUX/ArkOS-R3XS #130: Device type switch - R36H

Wirusko, 2025-04-24:
Hello, 
In the latest version of Arkos (03302025) after switching to the correct type of device (in my case R36H) I am not asked to confirm the "switch" by pressing A button. Screen just blinks or something. 

(Options - Device type - R36h) 
(switching to r36s works OK, r36h not working) 

That's the reason why shortucts for r36h are not working on this FW in my case. 
I've had to switch to the previous version, where it works.

AeolusUX, 2025-04-25:
Yes seems to be a bug on the uploaded builds. I would have to create a script to help you with when I get the time. If you're familiar with ssh on these devices you could also try and just add execute permissions to the script. It's at /opt/system/DeviceType/R36H.sh

AeolusUX, 2025-05-14:
This should be fixed with the latest update.

closed 2025-05-14 by AeolusUX