AeolusUX/ArkOS-R3XS #135: R36S No Access to Wifi since 04302025

ProfNuke, 2025-05-10:
Hi, I updated my R36S through Wifi and now i can not access the Wifi Settings within the Options menu. It tries to start and then goes back to the menu. Also shutted down and restarted the device. So I am now not able to update to further versions. Can anybody help how to fix that?

Kanaduchi, 2025-05-10:
Confirm

Wifi Settings within the Options menu don't work

ProfNuke, 2025-05-11:
also can start the update process via cable connected to smartphone with usb tethering

Kanaduchi, 2025-05-11:
> also can start the update process via cable connected to smartphone with usb tethering

This way is not possible for users who installed a wi-fi chip directly on the main board

ProfNuke, 2025-05-11:
i did not. i only using a usb-c plug wifi card

LeGIsTer, 2025-05-11:
Just make the /opt/system/Wifi.sh script executable!

PerryFabregas, 2025-05-12:
> Just make the /opt/system/Wifi.sh script executable!

But it is already executable, still not running 

[image]

LeGIsTer, 2025-05-12:
Connect SSH
sudo chmod +x /opt/system/Wifi.sh

ProfNuke, 2025-05-13:
Thank you. Will test it, but do you have any tutorial/video how to connect via ssh without network connection? (e.g. usb cable possible?)

ProfNuke, 2025-05-13:
perfect!!!! Thank you very much, it worked!!! yehaaw

closed 2025-05-15 by AeolusUX