AeolusUX/ArkOS-R3XS #139: Emulators not launching except psp portmaster nintendo ds

Ibrahim20082, 2025-05-15:
I have a big problem aftee latest update - every emulators except those that I listed had an errors. Firstly nothing was launching error was that es_systems.cfg file was empty. I filled it with one of file that was there es_sysems.cfg.update(dont remember witch).bak and so it launched but none of games launching, except psp emulqator nintendo ds and also 2 games from dreamcast but others not working. If problem with es_systems.cfg could you please put latest version od this file so I can paste there, if not couls you please say what to do, thanks

AeolusUX, 2025-05-15:
Sounds like your es_systems.cfg got corrupted. You don't need to edit the file just rename the .bak file of the last one.

Ibrahim20082, 2025-05-15:
> Sounds like your es_systems.cfg got corrupted. You don't need to edit the file just rename the .bak file of the last one.

I tried but it didnt work