AeolusUX/ArkOS-R3XS #149: Unable to use both micro SDs for ROMs in later version starting with ArkOS R3XS V2.0 (03302025)

AtumRa5, 2025-06-04:
Listen to me very closely, you closed it as the issue was "resolved" but it wasn't. I'm a computer scientist and I'm trying to help you and to also solve this bug. I did a fill more tests and this bug isn't present on ArkOS R3XS V2.0 (02092025).  Starting from version ArkOS R3XS V2.0 (03302025) onwards you've made some changes that broke this feature. I did try to reboot the console and this doesn't fix the problem. What is happening is this: On your OS I NEED to first click on "SWITCH TO SD2 FOR ROMS" option first, JUST THEN your OS unlocks the "READ FROM SD1 AND SD2 FOR ROMS" option for me to click, and there is where the bug happens and the “Emulationstation configuration files not found in /etc/emulationstation. Exiting” message appears. So I'm unable to do the switch so it looks for ROMs on both SD cards or to even revert it back to MAIN SD!! Even after rebooting the console! And the only way to revert back to MAIN SD is by formatting it again! I tried everything and this wasn't happening on version ArkOS R3XS V2.0 (02022025) and ArkOS R3XS V2.0 (02092025). You said you programmed it so in the main sd the file is renamed/removed? What?? Why? This breaks it! Permanently! Or maybe you forgot to make it reversible. The file should still be acessible because your OS is still trying to find it when I try to use the features it should have, the features present in your OPTIONS menu. So your ArkOS R3XS V2.0 (03302025) update broke these options, I'm telling you that these options aren't broken on older versions, so why are they broken now? You see? It is a bug introduced in the newer updates and others are having it as well, it is worthy of being investigated.

AeolusUX, 2025-06-04:
It's been that way since I first made this build. I setup these devices daily with the read from sd1 and sd2 for roms script and I never had your issue.

AtumRa5, 2025-06-04:
I'm using a 256GB Samsung EVO Plus 160MB/s XC I U3 A2 V30 white Micro SD card for SD2 and a 64Gb Sandisk Ultra 80MB/s XC I C10 gray and white SD card for SD1 (main system card).

AeolusUX, 2025-06-08:
Try this

1. Take the attached compressed file [arkosbackup.tar.gz](https://github.com/user-attachments/files/20646485/arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.
3. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.
4. Once completed, reboot the device.
7. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Tools, hit A, then scroll all the way down to es_systems_fix., hit A.
9. The unit will reboot again.
11. Try using the scripts again.

AeolusUX, 2025-06-16:
This should be accomodated on the latest update dropping in a few hours.

closed 2025-06-16 by AeolusUX

JoFonzi, 2025-06-17:
Thanks for the update.

Is there a process to doing this read from SD1 and SD2 option?
I clicked on the option and it completed but now I cant see any games from SD2 only SD1, so its the opposite now.