AeolusUX/ArkOS-R3XS #167: Support request for G80CA-MB V1.2 clone with BOE JC-T0350B64P-01G1 panel (DTB compatibility)

gacuna89, 2025-06-30:
Hi, first of all thank you for your awesome work on ArkOS!

I have a clone console that identifies as an R36S, but it's clearly a newer hardware variant not currently supported.

---

### 📌 Hardware details:

- **Board model:** G80CA-MB V1.2-20250422
- **Screen:** BOE 3.5" IPS - JC-T0350B64P-01G1 V1 (manufacture date: 02/23/2025)
- **RAM:** SK Hynix H9CCNNNBGTML-ARNUD (1 GB)
- **Original system:** Came with ArkOS 2.0 (04242024-AeUX), which worked, but I updated to a newer version and lost display and input.

---

### 🔍 What happened:

- After updating to ArkOS 2.0 (05312025), the screen went black and the red LED was blinking.
- I recovered partial functionality using a `.dtb` shared by Reddit user `chaoso85`.
- That `.dtb` fixed the screen — I now get image and boot sound.
- **However, the internal gamepad (joysticks and buttons) is not detected.**

---

### 🧪 What I've tried:

- Tested all DTBs available in the `R36S-DTB` repo (panels 1–6 and stock).
- Edited `extlinux.conf` to point to various `.dtb` files (renamed as `rf3536k3ka.dtb`).
- Replaced the DTB inside BOOT partition (ESP).
- Only the modified DTB from Reddit gives me video — but no gamepad functionality.

---

### 🙏 What I need:

- Help modifying or generating a working DTB for this specific board (G80CA-MB V1.2) and BOE panel.
- Or, if possible, include support for this variant in a future ArkOS update.
- I’m happy to help test builds or provide additional information, including photos or dumps.

---

### 🔗 Related Reddit threads:

- https://www.reddit.com/r/R36S/comments/1lb8fjs/so_i_knowingly_got_a_clone_for_tinkering_and_it/
- https://www.reddit.com/r/R36S/comments/1lb8fjs/comment/n02jizc/

These threads document the same hardware variant and partial DTB fixes shared by users like `chaoso85`.

Thanks again for your continued work!

gacuna89, 2025-06-30:
Here is the DTB file that gives me display but no input (just in case it helps): [[link]](https://github.com/AeolusUX/K36-DTB)
Panel 8

AeolusUX, 2025-07-01:
https://github.com/AeolusUX/K36-DTB does support it. It would not be supported on this image.