AeolusUX/ArkOS-R3XS #172: Viewport overrides no longer align

TheChristianGM, 2025-07-08:
After updating ArkOS to the latest version, I've noticed that all the custom viewport overrides I had for GB, GBC, and GBA no longer align with the physical screen layout.

Specifically, the vertical position (custom_viewport_y) appears to be offset — what used to be Y = 0 (which previously matched the top edge of the screen) is now visibly shifted down by 26 pixels.

For example, in GBA games using RetroArch and the mGBA core, I now have to set custom_viewport_y = 26 to restore the correct vertical alignment.

This issue is not limited to my unit — it also happens on my partner’s R36S, with identical results. It seems like something in the video output or RetroArch's default layout changed during the update.

AeolusUX, 2025-07-10:
I'm curious why this is considered an issue if adjustments can be made. Updates sometimes cause this, but it’s not intentional, and it seems to stem from RetroArch rather than the CFW developer. Could you explain why it’s a problem to begin with?

TheChristianGM, 2025-07-10:
I understand that this may not be an intentional change, and that it might stem from RetroArch itself. However, the current behavior is problematic because it breaks saved overlay configurations across multiple systems.

When updates shift the vertical screen alignment (like the 26px offset we've observed), all previously fine-tuned overlays and shaders become misaligned. This forces users to go through each system and readjust everything manually—core by core, system by system.

This is not just a cosmetic issue; for those of us who value accurate overlays and visual setups, it's a significant disruption. The expectation is that saved configurations should remain consistent across updates unless the user chooses to reset them.

I'd appreciate if this could be looked into for future stability. Even if the root cause is RetroArch, perhaps there could be a workaround or note in the changelog when such shifts occur.

AeolusUX, 2025-07-11:
I’m not addressing this issue due to time constraints and limited resources to test every system or configuration. Without knowing which version you updated from or to, it’s unclear how this issue arose, but you’ve already identified the workaround—adjusting the overlay by 26 pixels—so you need to handle this yourself. Any potential workaround would only be considered in a future update at our discretion.

To avoid issues:
- Back up your retroarch.cfg and overlay files before updating.
- Test updates on a separate SD card before applying them to your main setup. We only support SD card setups, not other configurations.

Always remember, updates in homebrew projects carry risks of unintended bugs, which is standard in this space and IT. If your setup is stable, avoid updating to prevent disruptions. This shift is a cosmetic issue, and adjusting configurations is part of the emulation process. I won’t be investigating further or providing warnings about such changes.