AeolusUX/ArkOS-R3XS #173: PortMaster have inaccessible folders

kai11, 2025-07-10:
I have a problem with PortMaster on R36S.
Im trying to run FridayNightFunkin port and it looks for Love env vars in /opt/system/tools/PortMaster/runtimes/love_11.5/love.txt.
This file is present on Boot partition, but it seems whole PortMaster folder got replaced with contents from tools dir on EASYROMS partition, which does not have this file or love runtime.
When I copy files from boot partition to EASYROMS, they deleted after reboot.

How this is supposed to work?
Why PortMaster exists in 2 places and why one replaces the other? Why my changes to PortMaster on EASYROMS are not preserved?

PS. Latest ArcOS-R36S 06302025-1 on new SD card, no modifications besides added 

Steps to reproduce:
- add friday night funkin zip from portmaster site to tools dir on easyroms partition
- install it with portmaster tool
- enable logs (DEBUG)
- try to run game
- check log for error

Easier way to reproduce:
- burn fresh image to new sd card
- check if /opt/system/tools/PortMaster/runtime folder present on boot partition
- insert card into device and boot it
- browse fsusing 351files from tools
- confirm there is no runtime folder

AeolusUX, 2025-07-10:
Why would the boot partition have Portmaster?

kai11, 2025-07-12:
I did not added portmaster to /boot - it was already there in ArcOs R36S image I wrote to my sdcard.
"Tools" dir refers to autoinstall directory of portmaster. 
EASYROMS /tools/PortMaster/autoinstall I think.

closed 2025-07-28 by kai11