AeolusUX/ArkOS-R3XS #178: screen 5 issue

torisar11, 2025-07-15:
I have r36s console with screen 5 but this is not clone, motherboard is original, i successfully install AmberElec but with ArkOS i have an issue, after install system don't react on any button, works only right analog stick if i press it diagonaly menu scrolling in random direction, also when system just boot i can see volume widget with volume turning down.

AeolusUX, 2025-07-15:
If you're using a Panel 5 device, you need to manually copy the Panel 5 files. Please refer to the instructions in the latest release notes.

Panel 5 devices are clones and are not officially supported by this image. No further assistance will be provided to Panel 5 users at this time.

closed 2025-07-15 by AeolusUX

torisar11, 2025-07-15:
Sorry but my device is not clone, i see many comparisons. It come with ArkOS originaly, 100% orig motherboard and other things that people love to compare. And main things, i try to use rom for clones and it never work, but official AmberELEC work perfectly, also ArkOS is installed without problems(i aso try to copy files manually) but when i in system my keys not work. And this is only in ArkOS

jindaxia, 2025-07-16:
I have the same device as yours