AeolusUX/ArkOS-R3XS #208: R36S V21 - OTG Port disabled - Wifi Off

Axythetaxi, 2025-08-22:
After installing version 06302025-1 to a new SD card and booting my R36S-V21 2024-12-18 I found that the OTG port would no longer function.

I've attempted multiple reboots, with the OTG Port occupied by a tested working USB wifi adapter, toggling the Wifi-Toggle script on and off and system reboots.

Having returned the Original SD card provided by seller. Running version 05312025. The OTG port regains functionality and operates the USB wifi adapter as expected. Wifi Toggle scripts also working when tested.

I have since tried, version 04302025 and found the same issue of OTG not functioning. I have been unable to download a fresh install of 05312025 from the Git as all links seem to have been updated/inactive so cannot verify if that build will work like version my stock SD card has installed.

I also tried the latest build of the R36S Plus (06302025-1) and found the OTG port would operate as intended but I now faced the issue of a higher resolution image on a smaller screen.

This has lead me to believe it is related to the unoccupied wifi slot available on this R36S revision board. I imagine it is considered active with a fresh install of Ark OS, with no way to toggle it in a way which enables the OTG port again. I understand that the built in Wifi cannot operate at the same time as the OTG port which is why I'm suspicious.

Let me know what information I can provide to help solve this issue. Thank you.

[image]
[image]

Axythetaxi, 2025-08-26:
Tested version 07312025 and unfortunately the additional of WiFi toggle 3.0 does not fix the issue like I had expected. Possibly missing drivers which are present in the R36S Plus package.

OTG does not enable after disabling "built in Wifi" 

Wifi status will not change regardless of enabling or disabling, with or without wifi adapter added to OTG port.
[image]
[image]

AeolusUX, 2025-08-28:
Not sure what build that is but maybe it's an internal build from the factory not what I distribute from this repo. They might have drivers for your dongle that we don't have unless we know what specific chipset your dongle is we won't know. My suggestion is get a supported one instead as even if you get the correct chipset it might take some time before I can add it on my builds.

Axythetaxi, 2025-08-28:
Thanks for the response.

I should have listed the dongle. Its a TL-WN725N

It works fine on my R36S V12 device running your most recent release/past releases.

I would like to point out that its the entire OTG port that is disabled when installing one of your builds. The OTG doesn't pick up USB Drives, USB controllers, or either of my USB WiFi dongles, through either of my working Usb C to A adapters.

Is there any reason the R36S Plus build would enable my OTG port to work as expected when the R36S builds would not?

Axythetaxi, 2025-08-30:
Currently testing my own work around.

I've installed the R36S Plus build 07312025 (AeUX) and replaced the boot.ini file with the one found on the Factory SD card.

The only differences in the boot.ini being "vt.global_cursor_default=0" added to the boot Args and the bootloader being changed "load mmc 1:1 ${dtb_loadaddr} rk3326-rg351mp-linux.dtb".

So I'm now just using the rk3326-rg351mp-linux.dtb provided with the R36S Plus image. and not it's original R36Plus version.

Screen resolution and aspect ratio looks fine. OTG ports working and Wifi Dongle connected.

Only other step I have taken is to set Device Type to R35S or R36S within the options menu.

An additional note, for some reason the SD card I am using did not have it's EASYROMS partition expanded on first boot.

VART1K, 2025-09-01:
I have the same issue. Tried USB Tethering and a WiFi dongle, both don't work with the latest ArkOS update on a R36S

Axythetaxi, 2025-09-01:
@VART1K 
> I have the same issue. Tried USB Tethering and a WiFi dongle, both don't work with the latest ArkOS update on a R36S

Is the OTG working for any other devices you plug in? Like a USB controller or USB Drive? Is it the same board revision as my own? I believe this is related to the pcb wifi spot more than anything.

@Julliejan-3 
> I have the exact same problem, none of the USB devices work with the OTG port unfortunately

Is this also a R36S V21 revision board from Boyhom? Have you attempted my work around to confirm. your OTG port os not damaged? Or is it working from with the Factory installed OS?

Jason3x, 2025-09-02:
Are you sure you have an original r36s and not a clone?

Julliejan-3, 2025-09-02:
> Are you sure you have an original r36s and not a clone?

Yeah im pretty sure its a original

Jason3x, 2025-09-02:
Have you tried replacing the dtb files with those from the original card?

Julliejan-3, 2025-09-02:
> Have you tried replacing the dtb files with those from the original card?

No not yet, can i just replace the dtb files? And leave the rest?

Axythetaxi, 2025-09-02:
@Jason3x 
> Have you tried replacing the dtb files with those from the original card?

To answer your question from my side, I have tried replacing each dtb file from the AeoluxUX release with the ones found on the Factory SD card.

Everything will boot as expected but the OTG will refuse to enable. This is what lead me to try the R36S Plus build as I know that has a built in wifi module. This fixed the OTG issue like I mention in my work around post.

All I can think of is the Wifi section on this PCB is not switching off when using one of AeoluxUX's releases, which is why the OTG is remaining blocked.

I was hoping your Wifi toggle would have fixed this in the most recent release but it may be being referenced differently on the pcb.

Axythetaxi, 2025-09-02:
@AeolusUX 
In short, yes.

I install the R36S Plus release onto the SD card and change boot.ini to look to the rk3326-rg351mp-linux.dtb

I do not need to transfer anything from the Factory card to make this work, the R36S Plus install seems to have the correct dtb already. But to be clear I did copy the boot.ini file from the Factory card as it includes the extra boot arg."vt.global_cursor_default=0"

Julliejan-3, 2025-09-03:
> Currently testing my own work around.
>
> I've installed the R36S Plus build 07312025 (AeUX) and replaced the boot.ini file with the one found on the Factory SD card.
>
> The only differences in the [...]

This worked for me, thanks!

samuelterra22, 2025-09-14:
Hello! I’m also experiencing this issue with the OTG port on the R36S using the latest ArkOS-R3XS release.

In my case, I tested with two SD cards:
	•	The original card with an earlier ArkOS version recognizes the Wi-Fi adapter through OTG without problems.
	•	The card with the latest build from this repository does not power the OTG port at all.

This confirms it’s not a hardware issue, but rather something in the kernel or configuration of this build.

I’d like to ask:
	1.	Is there a plan for a definitive fix to re-enable OTG support in upcoming releases?
	2.	Is there anything I could do to help — for example, testing kernel builds, adjusting boot parameters, or even contributing code/config changes?

Thanks for all the work on this project — I’d really like to contribute to solving this, since OTG support is essential for Wi-Fi modules and other peripherals.

AeolusUX, 2025-09-20:
> Hello! I’m also experiencing this issue with the OTG port on the R36S using the latest ArkOS-R3XS release.
>
> In my case, I tested with two SD cards: • The original card with an earlier ArkOS version [...]

1. No, because I cannot reproduce it with my devices so I can't figure out what's going wrong. My guess is some factory used a different spec on their manufacturing that made this issue. The original cards worked because that card has a patch somewhere not present in my builds. I haven't got a device with the same issue.
2. If you could provide me with an image without roms of the card you were using that has otg working for you I could probably cook up a build for people with this certain issue, or better if it doesn't do anything funny with the current build I might use it as a new base for future builds.

restrider91, 2025-09-20:
Good afternoon, my name is Jesús and I'm from Mallorca, Spain. I'm not a programmer, I'm just a technical graduate in microcomputer systems and networking (I don't know much about Linux). Over time, I've tried several ArkOS images on my R36S... and I've been using version 03302025 since it was released. With the latest update 07312025-1, I noticed, as have many others, that somehow, as mentioned before, even though the OTG port is working correctly, something is causing it to become unresponsive (whether using a Wi-Fi dongle or, as I do, using the RTL8811CU 2.4 GHz/5.8 GHz Wi-Fi module).  Before I gave up on getting the Wi-Fi working, Jason's script would cause problems when I disabled and then re-enabled Wi-Fi; neither the dongle nor the 8811CU would work after that. This made me wonder if the wrong .dtb file was being specified in the boot.ini file.  Interestingly, that wasn't the case; the path to the .dtb file is correct (whether in the boot.ini file from 03302025, 05252025, or 07312025-1).  All of this sparked my curiosity, so I asked ChatGPT what could cause the OTG port to appear as "dead" when it's actually functional, and why the Wi-Fi (whether using a dongle or the 8811CU adapter) wouldn't activate.  ChatGPT mentioned something about addressing, and knowing that was correct, I asked if there could be a problem with the text lines in the .dts files.  I then noticed that in most of these files (which were modified 3 months ago), there are many lines with "status=disabled". In fact, ChatGPT said that this could be the reason why neither the Wi-Fi nor the port are working.  Therefore, I think it would be a good idea to review all the lines in those files and change them to "okay". Here are the lines or sections in the .dts files that I recommend checking (although I think there are more disabled entries that I haven't listed here): usb@ff340000, usb@ff350000, serial@ff160000 to serial@ff178000, usb2-phy@100/otg-port, dwmmc, nandc, sfc, ethernet@ff360000, i2s@ff080000. I don't know if this will help solve the problem Aeolux is having, but I just wanted to contribute my ideas. Knowing that it's not related to boot.ini or drivers, this is what I could think of as a possible solution.

mexxx77, 2025-09-24:
Here is my original BOOT folder of my device in case it can help you identify what's different.

[https://mega.nz/folder/v0YE0QhJ#0fIiULwExkRqsWaXCuEptQ](url)

niokofast, 2025-09-25:
Almost same issue here:

Installed a RTL8723BU Wifi module internally on the dedicated WIFI Slot.
- Immediately works. Even after reboot.

When I use Jason's WIFI switcher to disable it. It will never start again.

- Reflashed an SD Card --> works again until disabled once.
The OTG Port sometimes works, but sometimes doesn't. I have not found out how to get it to work, but after many reboots it sometimes works, and my wired USB-C Ethernet Adapter is usable. LSUSB won't list the WIFI Card when accessing this way, but I suspect that this is by design.

It seems to me, that the WIFI Toggle is able to disable the module, but not to reanable. 
Does anyone know how the switching between the modes works? Is this via a GPIO?

I now followed @Axythetaxi s approach on using the R36S Plus Image. I did however just copy all the files of the boot partition from my "current" SD Card, because I only got a black screen with it. Unfortunately, it behaves the same as I described above.

niokofast, 2025-09-26:
@Jason3x 

Jason! That's a surprise - thanks for your clarification + suggestions, much apprechiated.

Funny enough, I stumbled over the "preferred modules", but then for whatever reason didn't follow up on it. 
- tried your updated script --> no change 
- added all the blocklist.conf devices from the end to the preferred modules --> no change
- I've removed all the devices from blacklist.conf --> no change

Didn't have much time yesterday, but will dig further.

Do you know how to certainly get the device name, that would need to be put into preferred devices?

niokofast, 2025-09-26:
@Jason3x 
Ah, now I understand, your script adds the device that it disables to the blacklist.conf, but does not remove it, if the device is not in the "preferred moduls"? Correct?
When I checked the modules, I did it when the wifi was working, so the affected module was not yet in the blacklist.conf

The module listed in blacklist.conf after deactivating wifi is: 8723bu

I'm at work, and don't have a possibility to access the console trough SSH, so I wasn't able to add the module to the preferred list. 
I will test ASAP and get back to you.

Thanks again.

xthndr, 2025-09-26:
Same issue. Did you still need an image? I can try and create one from factory sd. V21 board with the WiFi pcb space.

niokofast, 2025-09-27:
@Jason3x 

Sorry, took a while..

V 3.0 with added Module to preferred list works without any issues.
V 3.2 Works, but not after the console is restarted. So if I disable, restart, Enable --> Does not work. IT looks like the Script crashes, it it goes from "Enabling Wifi" to a black screen, and then back into the advanced menu. It shows "Enabling..." for a fraction of a second. I cannot read it.

V 3.2 with added Module to preferred makes no difference.

For me, I'm happy with 3.0 now working, so I don't bother too much, but if you want me to do further tests, just let me know.
Thanks again for your help.

Jason3x, 2025-09-27:
@niokofast ,

Thank you so much for these tests, and thank you very much for the coffee 😉. I only test my scripts on a console because that's all I have lol. So I'm going to put version 3.0 back on my git. If you want to test something else, I suggest [Session_Recall](https://github.com/user-attachments/files/22576361/Session_Recall.sh)

xthndr, 2025-09-28:
Using the plus image on my v21 board worked as a workaround to enable WiFi. I did as instructed above and swapped the boot.ini and dtb files. Only issue I have is the initial boot screen is garbled. Assuming it uses a different file the initial boot and I just don’t replace that one or something

Jason3x, 2025-09-29:
@niokofast 

I reverted the script to v3.3, so there's no need to change the preferred modules, they'll be automatically blacklisted and unblacklisted. 
Can you confirm that?

[Wifi-toggle.sh](https://github.com/user-attachments/files/22604513/Wifi-toggle.sh)

niokofast, 2025-09-30:
@jimmy213 
True, and sorry for that, I was under the impression, that it is related, because OTG wasn't working anymore after switching. Turned out that it wasn't related.

nizarom, 2025-10-12:
I’ve got the same issue with R36S, the stock SD card the OTG port work fine but when I installed ArkOS 2.0 (07312025-1) the OTG port has been completely dead even for USB Key.
Update: when i click the start button the OTG port work for a while and stop after

abit11w, 2025-10-18:
same unfortunate issue

ZmenszonySeba, 2025-10-18:
> same unfortunate issue

Unfortunately, no one here will help you — I’ve already lost hope that anyone will ever fix it.

nizarom, 2025-10-21:
Any solution guys?

bmcharg, 2025-10-22:
I have the same problem with a v22 board. If I boot with a clean install of AeUX, the OTG port is dead. Plugging in a keyboard or WiFi dongle yields no results. If I boot the same device with the stock card supplied with it, both the keyboard and dongle work so I can confirm they are compatible.

 I've managed to take one step forward but two steps back in some exploratory attempts at fixes. Here's what I've done;

1. Flash AeUX onto a new card
2. Boot the device with the new card (to complete initial setup)
3. Copy `rk3326-rg351mp-linux.dtb` and `gameconsole-r36s.dtb` from the stock card supplied with the console to the AeUX card
4. Create two empty files on the root of the AeUX card, `nopanelchooser` and `unsetpanelchooser` (just to make sure that the panel chooser doesn't interfere) My device is a Panel 4 
5. For further security, rename `ScreenFiles` to `ScreenFiles_backup`
6. Edit the `boot.ini` file on the AeUX card changing line `load mmc 1:1 ${dtb_loadaddr} "${PanelPathSlash}rk3326-r35s-linux.dtb"` to `load mmc 1:1 ${dtb_loadaddr} "rk3326-rg351mp-linux.dtb"`
7. Boot the device using this card

When then booting the device I think get a configuration screen that asks me to define the key inputs again (I presume because it detects the change in dtb file) I can define _most_ keys, but this is where I then hit my first problem. The Up, Down, Left and Right positions of both anolog sticks isn't picked up :(

If I ignore those for now and complete the configuration, the device boots fine and the OTG port works! I can navigate the UI fine, but the analog sticks don't work. So then I tried connecting to my WiFi and hit problem two. I can navigate to Options > WiFI fine using the D-Pad and then pressing A and then choose **Connect to new WiFi connection** which then sees my SSIDs! However, when I try to enter the key, the buttons are all messed up (despite working fine in the UI).  Up and Down on the d-pad navigate Left and Right, and L2 and R2 are Up and Down. To be sure I ran all these steps twice, just in case I'd defined them wrong, but it feels like the settings are being honoured differently in the shell rather than in the UI.

If I work around this error and input the wifi key it all connects fine and I have internet access!

Hopefully this makes sense to someone above my pay grade and helps get to a working solution that gets OTG working, the analog sticks registering and the controls being mapped correctly in the shell screens!

Feels like a solution could be close ..... If anyone else wants to try the same steps, here are the dtb files I'm using from my stock card in the steps above. Again to confirm, this is an R363-v22 with a Panel 4 screen

[dtb-files.zip](https://github.com/user-attachments/files/23058133/dtb-files.zip)

Axythetaxi, 2025-10-22:
@bmcharg This is as far as I also got when testing swaps of dtb files with @Jason3x 

I've been unable to dedicate any more of my time but I am checking the thread.

I tried a method of replacing the image file with one Jason provided me to fix the remapping but it just killed the install and I had to stop there.

bmcharg, 2025-10-22:
ok, I've stuck with it and think I may have something ....

I’ve confirmed that the R36S OTG 5 V line isn’t enabled when using the rk3326-rg351mp-linux.dtb shipped with the AeUX build. The issue is that the DTB lacks a reference to the vcc_host regulator node that drives the USB VBUS power rail.

Here's the steps I took to patch it and get it up and running;

**1. Decompile both DTBs from the bundled sd card BOOT partition**

```
dtc -I dtb -O dts -o mp.dts rk3326-rg351mp-linux.dtb
dtc -I dtb -O dts -o r36s.dts gameconsole-r36s.dtb
```

**2. Copy the R36S regulator block into the MP DTS**

```
vcc_host: vcc_host {
    compatible = "regulator-fixed";
    regulator-name = "vcc_host";
    gpio = <0x5c 0x0f 0x00>;
    enable-active-high;
    regulator-always-on;
    vin-supply = <0x64>;
};
```

**3. Inside the USB2 OTG node (`usb@ff300000 with dr_mode = "otg"`) add**

```
vbus-supply = <&vcc_host>;
```

**4. Recompile and replace the DTB**

```
dtc -I dts -O dtb -o rk3326-rg351mp-linux.dtb mp.dts
```

This triggers the OTG 5 V power on boot, so WiFi works, and the analog sticks and button mappings are correct.

If anyone else wants to test, you can extract the two files in [boot-partition-patch.zip](https://github.com/user-attachments/files/23063181/boot-partition-patch.zip) to your BOOT partition and give it a go. 

For the devs, here is the decompiled and modified DTB file [mp.dts.zip](https://github.com/user-attachments/files/23063217/mp.dts.zip) with the changes being lines 11-18 and line 1764.

Interested to hear how you guys get on @Axythetaxi and @nizarom

getpanoramic, 2025-10-22:
> Fantastic news! Just for ease for anyone else that comes across this thread, you can just extract this zip over your BOOT partition and replace all files.
> 
> [patched-boot-partition.zip](https://github.com/user-attachments/files/23063504/patched-boot-partition.zip)
> 
> Obviously I've only tested this on my hardware, an R36S-v22 Panel 4, but hopefully there's enough here that someone with a better understanding of the system as a whole can integrate it into the main branch in a way that is compatible with all the hardware combos!

I've just given it a test, but it doesn't seem to work. I also have an R36S v22 Panel 4.

When I use my SD card with the stock OS, my Wi-Fi dongle/OTG works. Not sure how I can debug this.

bmcharg, 2025-10-22:
hmm, do you have different dtb files on your stock card? could you send them over and I can have a quick look. and when you say it doesn't work, what happens?

bmcharg, 2025-10-23:
That's great news!

Axythetaxi, 2025-10-25:
> Fantastic news! Just for ease for anyone else that comes across this thread, you can just extract this zip over your BOOT partition and replace all files. [...]

@bmcharg I can confirm that this is also working for me on my R36S-V21 2024-12-18. OTG port is enabled.

Thank you for solving this issue. I hope @AeolusUX can find a way to incorporate the changes you've made into an upcoming build.

My only question, is the boot partition zip you provided meant to be replaced after first boot or before first boot? I used it prior to first boot and it worked fine but I'm yet to see if there were any adverse effects.

AeolusUX, 2025-10-28:
Not sure if this affects the old v21 boards so I have to add this new dtb as a seperate thing for now.

JeffAllison, 2025-10-28:
Works on my V21 but not my V22.

jrwells2, 2025-11-01:
> Fantastic news! Just for ease for anyone else that comes across this thread, you can just extract this zip over your BOOT partition and replace all files. [...]

For this to work what buold am I patching? The plus build? Clean arkos? Aux version?

wekii, 2025-11-02:
Guys, **you only need boot.ini** from _patched-boot-partition.zip_, nothing else !!

_rk3326-rg351mp-linux.dtb_ from _patched-boot-partition.zip_ is not modified !!

But **it must be fresh installation** of ArkOS R3XS V2.0 (07312025-1) on SD card before first boot.

These are just those two things in modified boot.ini that are different from the original boot.ini file:

[image]

I don't understand why @bmcharg told us to replace everything from _patched-boot-partition.zip_?!
It just doesn't make sense.

bmcharg, 2025-11-03:
> Guys, **you only need _boot.ini_ and _rk3326-rg351mp-linux.dtb_** from _patched-boot-partition.zip_, nothing else !!
> 
> But **it must be fresh installation** of ArkOS R3XS V2.0 (07312025-1) on SD card before first boot.
> 
> These are just those two things in modified boot.ini that are different from the original boot.ini file:
> 
> <img alt="Image" width="1729" height="427" src="https://private-user-images.githubusercontent.com/8656086/508722452-39ffa651-59fa-4456-9178-95a098a8b8d2.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjIxNjg5MzEsIm5iZiI6MTc2MjE2ODYzMSwicGF0aCI6Ii84NjU2MDg2LzUwODcyMjQ1Mi0zOWZmYTY1MS01OWZhLTQ0NTYtOTE3OC05NWEwOThhOGI4ZDIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MTEwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTExMDNUMTExNzExWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjQzNWY2MDAwZjE5NzNjODVhZmIzYzY0NDdhOTZlZDk4Y2ZjY2I0OGZlZGMzYjMwNzE2OWE5NjkzMzIyYmVlMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.WyghL0TZAWY1oLWX2xWO3mbVxWObcHZnJf-48JVlXGs">

Thanks @wekii 

I'd already supplied a separate zip with just those two files here https://github.com/AeolusUX/ArkOS-R3XS/issues/208#issuecomment-3434409153. The `patched-boot-partition.zip` you're referring to was a full zip I made for anyone that wants to just replace the _entire_ BOOT partition for simplicity. if you use the first patch that will only do the two files you're referring two (which are indeed the only two that need replaced)

wekii, 2025-11-04:
Good question, I'd like to know that too 😁

AnotherMuggle, 2025-11-04:
Patch also working for me on V22 Panel 4.  Thank you!  

Do we need to be careful about updating in the future or will this be integrated into the main build?