AeolusUX/ArkOS-R3XS #212: R36S V2.0 (07312025) error PPSSPP 1.19.3

hoanganh2020, 2025-08-28:
I upgraded and found all PSP games are broken when using PPSSPP 1.19.3

CenGo86, 2025-08-30:
Same here after update no psp game works

AeolusUX, 2025-08-30:
Update to what? I need more details.

CenGo86, 2025-08-30:
I´m on 07312025 after this my ppsspp dont work anymore on my R36S. 

I have to select "Standalone 2021" then it works. Auto no game start´s.

AeolusUX, 2025-08-30:
> I have the same issue as the other commenters, PPSSPP is not working when standalon is selected. standalone 2021 has to be selected for any of them to launch. running 07312025.

again update to the latest

hoanganh2020, 2025-08-30:
> > I have the same issue as the other commenters, PPSSPP is not working when standalon is selected. standalone 2021 has to be selected for any of them to launch. running 07312025.
> 
> again update to the latest

I have downloaded and updated many times. This PSP game system cannot be played when you upgrade PPSSPP to version 1.19.3

CenGo86, 2025-08-30:
Where can i find the 07312025-1 ?

willwills90, 2025-08-30:
> > > > I have the same issue as the other commenters, PPSSPP is not working when standalon is selected. standalone 2021 has to be selected for any of them to launch. running 07312025.
> > > 
> > > 
> > > again update to the latest
> > 
> > 
> > I have downloaded and updated many times. This PSP game system cannot be played when you upgrade PPSSPP to version 1.19.3
> 
> That is not the latest. Latest is 07312025-1

07312025-1 is not showing on the releases page for me at least