AeolusUX/ArkOS-R3XS #217: R36S - OTG and WiFi Dongle not working

VART1K, 2025-09-01:
[image]

I have bought a WiFi dongle that is supposed to be compatible with R36S, judging by the reviews and that this is a designated WiFi dongle that is sold by the official R36S console seller on AliExpress.

https://www.aliexpress.com/item/1005009040154737.html

It is a 802.11n RTL8188FTV WiFi Dongle
There is even a tutorial on how to use it with the ArkOS, by installing some drivers:

https://www.teamxnl.com/installing-cheap-wifi-on-your-r36s-or-r36h/

But I have encountered an error that says that the RTL8188FTV driver is already install and that it cannot install it, so the Wifi should work.

The guy that made the driver even states that it is not compatible with this ArkOS version.
https://youtu.be/3Ka4UdghNhQ?si=Hcu-oT24_zcfbKNM&t=641

Is there any update coming to fix this OTG WiFi issue?

Thank you!

AeolusUX, 2025-09-06:
To know which chipset you actually have, plug your device into your computer and check directly. Don’t rely on product listings since sellers can easily ship you a random chipset Wi-Fi dongle and you’ll be stuck with something that isn’t compatible.

There won’t be any updates to fix unsupported Wi-Fi dongles. If you want it to work, you’ll need to buy a dongle with a confirmed compatible chipset.

You’re also misunderstanding that video. It clearly says the drivers are already included in the community image and you don’t need to reinstall them. If your RTL8188FTV dongle isn’t working, the problem is either with your OTG port, your OTG adapter, or the dongle itself.

closed 2025-09-06 by AeolusUX

f22raptorroland, 2025-11-08:
I have the same white R36S, and bought the same Aliexpress FTV listed dongle.

I also have the same issue.
But I still have the original SD card it came with, and that is working (with speaker noise and all)

But in today's release and the latest few I tried, the above-pictured effect is what I get. :( 

With the factory SD card, the speaker is already noisy during boot, but in this case, it's totally silent.
The Wi-Fi toggle reports it to be Off, and it stays like that no matter what... 

I compared the /etc/modprobe.d/blacklist files are totally identical.

Do you have some recommendations on what to try and check? @AeolusUX 
Considering I have the working and non-working systems. (Although I think the factory one is not running this ArkOs)

Is it possible to remove the RTL drive in the system and install the one from the video to test?

USB is reported as ETV, not FTV(as in aliexpress)
```
        +++++++++++++++++ Device Information ++++++++++++++++++
Friendly Name            : Realtek RTL8188ETV Wireless LAN 802.11n USB 2.0 Network Adapter
Device Description       : Realtek RTL8188ETV Wireless LAN 802.11n USB 2.0 Network Adapter
Device Path 1            : \\?\USB#VID_0BDA&PID_0179#04BA362BA7E9#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Device Path 2            : \\?\USB#VID_0BDA&PID_0179#04BA362BA7E9#{435b6226-1dcc-43b3-887e-217dbaa27ba3}
Device Path 3            : \\?\USB#VID_0BDA&PID_0179#04BA362BA7E9#{ad498944-762f-11d0-8dcb-00c04fc3358c}\{6196e44f-eb57-4549-8bdc-1e118495a782} (GUID_NDIS_LAN_CLASS)
Device Path 4            : \\?\USB#VID_0BDA&PID_0179#04BA362BA7E9#{cac88484-7515-4c03-82e6-71a87abac361} (GUID_DEVINTERFACE_NET)
Kernel Name              : \Device\USBPDO-20
Device ID                : USB\VID_0BDA&PID_0179\04BA362BA7E9
Hardware IDs             : USB\VID_0BDA&PID_0179&REV_0000 USB\VID_0BDA&PID_0179
Driver KeyName           : {4d36e972-e325-11ce-bfc1-08002be10318}\0015 (GUID_DEVCLASS_NET)
Driver                   : \SystemRoot\System32\drivers\rtwlanu.sys (Version: 1030.38.712.2019  Date: 2024-04-01)
Driver Inf               : C:\WINDOWS\inf\netrtwlanu.inf
Legacy BusType           : PNPBus
Class                    : Net
Class GUID               : {4d36e972-e325-11ce-bfc1-08002be10318} (GUID_DEVCLASS_NET)
Service                  : RtlWlanu
Enumerator               : USB
Location Info            : Port_#0003.Hub_#0010
Location IDs             : PCIROOT(0)#PCI(0102)#PCI(0000)#PCI(0800)#PCI(0003)#USBROOT(0)#USB(4)#USB(3)#USB(4)#USB(3)#USB(3), ACPI(_SB_)#ACPI(PCI0)#ACPI(BXBR)#ACPI(BYUP)#ACPI(BYD8)#ACPI(XHC0)#ACPI(RHUB)#ACPI(PRT4)#USB(3)#USB(4)#USB(3)#USB(3)
Container ID             : {0df60ef3-661d-536e-a743-aa6271447291}
Manufacturer Info        : Realtek Semiconductor Corp.
Capabilities             : 0x14 (Removable, UniqueID)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Upper Filters            : vwifibus
Power State              : D0 (supported: D0, D1, D2, D3, wake from D0, wake from D1, wake from D2)
 Child Device 1          : Microsoft Wi-Fi Direct Virtual Adapter
  Device Path 1          : \\?\{5d624f94-8850-40c3-a3fa-a4fd2080baf3}#vwifimp_wfd#d&30c83530&0&11#{ad498944-762f-11d0-8dcb-00c04fc3358c}\{f3ae1970-5b52-4e2c-a12a-6e39b0ce6363} (GUID_NDIS_LAN_CLASS)
  Device Path 2          : \\?\{5d624f94-8850-40c3-a3fa-a4fd2080baf3}#vwifimp_wfd#d&30c83530&0&11#{cac88484-7515-4c03-82e6-71a87abac361} (GUID_DEVINTERFACE_NET)
  Kernel Name            : \Device\000000b7
  Device ID              : {5D624F94-8850-40C3-A3FA-A4FD2080BAF3}\VWIFIMP_WFD\D&30C83530&0&11
  Class                  : Net
  Driver KeyName         : {4d36e972-e325-11ce-bfc1-08002be10318}\0016 (GUID_DEVCLASS_NET)
  Service                : vwifimp
  Location               : VWiFi Bus 0
```

f22raptorroland, 2025-11-08:
[bmcharg](https://github.com/bmcharg) Posted a fixed partition. That solved it for me. Maybe it's an OTG issue not a Wi-Fi issue...

Just download the file and extract it over my boot partition.
Source Issue:(https://github.com/AeolusUX/ArkOS-R3XS/issues/208#issuecomment-3434478371)

AeolusUX, 2025-11-09:
> I have the same white R36S, and bought the same Aliexpress FTV listed dongle.
>
> I also have the same issue. But I still have the original SD card it came with, and that is working (with speaker noise [...]

Use Panel 4 V22 from https://github.com/AeolusUX/R36S-DTB