AeolusUX/ArkOS-R3XS #221: pr-evdev-uInput (1/1) not configured - device puts itself to sleep every 5 minutes regardless of Menu > Advanced Configuration > Auto suspend timeount > Off

etyrnal, 2025-09-09:
pr-evdev-uInput (1/1) not configured 

device puts itself to sleep every 5 minutes regardless of Menu > Advanced Configuration > Auto suspend timeount > Off

Have tried Toggling it on and off also.

R36S plus running ArkOS 2.0 latest

How can i fix this.  Is there a flat file i can manually edit to fix this?

Or a keybind that's broken that can be set or whatever?  There are several others mentioning this in reddit.

AeolusUX, 2025-09-16:
Never experienced this. What game are you playing and what emulator?
What card are you using?

AeolusUX, 2025-09-20:
> Every game. Any emulator. Card is new SanDisk 256GB High Endurance.

Are you sure? because some games don't use retroarch and uses standalone emulators which don't have this error. I suggest you be specific so I can reproduce it on my end to see if it really is an issue with the build or an isolated case on your install or hardware.