AeolusUX/ArkOS-R3XS #229: RLT8821CU Needs Mode Switch

showmikahmed4, 2025-09-26:
I am facing a issue where the USB device mounts as Disk in ArkOS, but works after I use USB Mode Switch. But this needs to be done in every single boot. The chipset is - RLT8821CU, which supports both wifi and bluetooth. I am giving a picture of the device, device details before and after the usb mode switch. It would be very helpful if you could add a permanent script to boot the device in Wifi mode on startup also a bluetooth script to connect controllers, keyboards, audio devices etc. Thank you in advance!

[image]
[image]
[image]
[image]

showmikahmed4, 2025-09-28:
> Hey, just a quick update on my side:
> 
> I’ve got it working now – in my case, the issue was the onboard WiFi module.
> I desoldered it, and after that, OTG started working, and the 8821CU dongle works without any problems.
> 
> I never had the usb_modeswitch issue you mentioned, where you need to switch modes on every boot – that just didn’t happen on my end.
> 
> Maybe it could help if you try flashing the R36S Plus firmware, and then copy over your current boot.ini and the correct panel.dtb from your boot folder.
> Might fix the mode-switching issue you're having.
> 
> Also, once you have the firmware installed, go to Options -> Device Type -> R35S or R36S to make sure the device is set correctly.

Thanks, I will try.

KalleTheodor, 2025-09-28:
could you please give me some feedback if it worked for you?  
I would be really interested to know whether the solution fixed the issue on your side.
 
thx

showmikahmed4, 2025-09-30:
> do you copy .dtb and boot.ini?

Yes. But I will give it another shot and let you know.

lcdyk0517, 2025-10-04:
[arkos functional script.sh](https://github.com/user-attachments/files/22691509/arkos.functional.script.sh)

The script I have updated and can already be persisted

showmikahmed4, 2025-10-04:
> [arkos functional script.sh](https://github.com/user-attachments/files/22691509/arkos.functional.script.sh)
> 
> The script I have updated and can already be persisted
> 

Can the usb mode switch be automated in this new script? And does bluetooth work now? I had an idea where the devices which already had bluetooth support in ArkOS, we can extract the bluetooth script from that and use in combination with your script.

lcdyk0517, 2025-10-04:
I have tested the xinput controller, but there is no driver. The dinput can be driven, and you need to add the mapping of es_input yourself. I have also tested the Bluetooth headset and it works normally.

lcdyk0517, 2025-10-04:
The script should integrate Bluetooth scanning and connection. You can test it yourself.

showmikahmed4, 2025-10-04:
Okay thank you, I will try and let you know.

showmikahmed4, 2025-10-05:
> The script should integrate Bluetooth scanning and connection. You can test it yourself.
> 

The persistent function works! Thank you so much. But I can't connect to any of my Bluetooth speakers. It can scan but it never connects. It just jumps back to the home screen. I have BlueZ installed.

KalleTheodor, 2025-10-05:
Enable SSH (Remote → Enable Remote) and connect using PuTTY.

Download the firmware files from here:
 [rtl8821c_firmware.rar (MediaFire)](https://www.mediafire.com/file/rrqr35h6kid4qpu/rtl8821c_firmware.rar/file)

I couldn’t upload them directly here, so I’ve shared them externally.

Extract the two files rtl8821c_fw.bin and rtl8821c_config.bin, then copy them to
/lib/firmware/rtl_bt/ (or possibly /usr/lib/firmware/rtl_bt/) —
I’m not 100% sure about the exact path, so please double-check.

Restart the Bluetooth service or reboot the device.

After that, the stick should be recognized.

KalleTheodor, 2025-10-06:
@lcdyk0517 @showmikahmed4 quick update from my side,

I now have a fresh R36S non Plus unit and I soldered the E103 RTL8811CU board directly onto the mainboard as I mentioned before. Running the latest ArkOS version WiFi was detected immediately and the device is correctly recognized as NIC not as a disk

Regarding the USB WiFi dongle I strongly advise against using it. At around 27 percent battery (R366S Plus) WiFi cuts out likely due to unstable voltage conversion from 5 volts to 3.3 volts which causes reliability issues.

At the moment I only use a Bluetooth keyboard via the soldered USB dongle!

Please give me some time as I just became a father!

More test results and photos will follow soon ;-).

Cheers
Kalle

showmikahmed4, 2025-10-25:
> The script should integrate Bluetooth scanning and connection. You can test it yourself.
> 

I have tested bluetooth, but can only scan and can't connect anything. Could you share if you updated the script or installed anything else?

It would be very helpful if you could share the script as a separate post.

AeolusUX, 2025-11-30:
Should be catered with the addition of ArkManager from lcdyk0517

closed 2025-11-30 by AeolusUX